rbei-etas/busmaster #772: the issue regarding the patch usage for CStringt.h

tolstiybh, 2015-02-17:
Hello.

Currently I am working on the Visual Studio 2013 ultimate. I had tried to compile the solution, however I received the error regarding cstringt.h library. Then I followed recommendations concerning the WinDDK patch, but when I launched this patch, it got a notification: "patch" is not external or internal command of the running programor package file. For the continue press any key.
 I assume that patch's changes hadn't been applied. Therefore, what this notification mean and what should I do in order to properly execute this patch? 

Thanks in advance)

mgttlinger, 2015-04-22:
I believe your issue is that your installation of git is not where the script expects it to be and therefore the `patch` command is not available.

tolstiybh, 2015-04-22:
Thank you very much for your reply.
You're absolutely right. I have done it. 
--
Отправлено из мобильной Почты Mail.Ru