rbei-etas/busmaster #773: J1939 Simulation

gt-morgott, 2015-02-18:
Hello.

I tried to send a J1939 wih the J1939-Simulation but it failed. After some research i detected that there is a problem with the struct STJ1939_MSG und their bitfields. The corresponding data don't transmit in the right way. I think the reason is based in the construction of the bitfields. I changed the structs as shown afterwards. Similarly i modified the function g_unSendMsg_1939 to parse the data of the modified structures in the right way. I hope the problem can be reproduced.

The following code shows my changes:

/*** Sources\DataTypes\J1939_DataTypes.h ***/
/*** Sources/Application/SimulatedSystems/include/Wrapper_J1939.h ***/
/* J1939 PGN structure */
/* J1939 Extended 29 bit ID */
typedef struct tagSTRUCT_29_BIT_ID_mod
{
	BYTE m_bySrcAddress : 8; // Sender node address.
	BYTE m_byPDU_Specific : 8; /* PDU Specific (PS), Destination address if 0 <= PF <= 239.
							   Group Extension (GE)if 240 <= PF <= 255. */
	BYTE m_byPDU_Format : 8; // PDU Format (PF), Indicates Peer_2_peer or Broadcast.
	BYTE m_byDataPage : 1; // DataPage (DP), Set to 0 currently.
	BYTE m_byReserved : 1; // Reserved, Set to 0.
	BYTE m_byPriority : 3;
BYTE: 3;

} STRUCT_29_BIT_ID_mod;

/* J1939 message properties */
typedef struct tagSTJ1939_MSG_PROPERTIES_mod
{
	UINT64					m_un64TimeStamp; // Timestamp.
	BYTE					m_byChannel;     // Channel number.
	EJ1939_MSG_TYPE			m_eType;
	EDIRECTION				m_eDirection;    // DIR_TX, DIR_RX
	STRUCT_29_BIT_ID_mod	m_uExtendedID;   // 29 bit extended ID

} STJ1939_MSG_PROPERTIES_mod;

/* J1939 message structure */
typedef struct tagSTJ1939_MSG_mod
{
	//public:
	/* Public member variables */

	static int  m_nSortField;
	static int  m_nMFactor;     // Multiplication factor
	STJ1939_MSG_PROPERTIES_mod	m_sMsgProperties;   /* J1939 message properties. (29 bit Extended ID, Timestamp,
													Channel, Type, Direction) */
	UINT						m_unDLC;            // Data length. 0 <= m_unDLC <= 1785 bytes.
	BYTE*						m_pbyData;          // J1939 message data.

} STJ1939_MSG_mod;
/*** ***/

/*** Sources\NodeSimEx\AppServicesJ1939.cpp ***/
UINT g_unSendMsg_J1939(STJ1939_MSG_mod* psTxMsg_mod, HMODULE hModule)
{
	/* copy parameters of modified structure, because original structure did not transmit parameters properly */
	STJ1939_MSG *psTxMsg;
	psTxMsg = new STJ1939_MSG;
	psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_uPGN.m_sPGN.m_byReserved = psTxMsg_mod->m_sMsgProperties.m_uExtendedID.m_byReserved;
	psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_uPGN.m_sPGN.m_byDataPage = psTxMsg_mod->m_sMsgProperties.m_uExtendedID.m_byDataPage;
	psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_uPGN.m_sPGN.m_byPDU_Specific = psTxMsg_mod->m_sMsgProperties.m_uExtendedID.m_byPDU_Specific;
	psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_uPGN.m_sPGN.m_byPDU_Format = psTxMsg_mod->m_sMsgProperties.m_uExtendedID.m_byPDU_Format;
	psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_uPGN.m_sPGN.m_byPriority = psTxMsg_mod->m_sMsgProperties.m_uExtendedID.m_byPriority;
	psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_bySrcAddress = psTxMsg_mod->m_sMsgProperties.m_uExtendedID.m_bySrcAddress;
	psTxMsg->m_sMsgProperties.m_byChannel = psTxMsg_mod->m_sMsgProperties.m_byChannel;
	psTxMsg->m_sMsgProperties.m_eDirection = psTxMsg_mod->m_sMsgProperties.m_eDirection;
	psTxMsg->m_sMsgProperties.m_eType = psTxMsg_mod->m_sMsgProperties.m_eType;
	psTxMsg->m_sMsgProperties.m_un64TimeStamp = psTxMsg_mod->m_sMsgProperties.m_un64TimeStamp;
	//psTxMsg->m_nMFactor = psTxMsg_mod->m_nMFactor;
	//psTxMsg->m_nSortField = psTxMsg_mod->m_nSortField;
	psTxMsg->m_pbyData = psTxMsg_mod->m_pbyData;
	psTxMsg->m_unDLC = psTxMsg_mod->m_unDLC;

    UINT Return = 0;
    VALIDATE_POINTER_RETURN_VAL(psTxMsg, Return);
    CExecuteFunc* pmCEexecuteFunc =
        CExecuteManager::ouGetExecuteManager(J1939).pmGetNodeObject(hModule);

    if (pmCEexecuteFunc != nullptr)
    {
        BOOL bMsgTxFlag = pmCEexecuteFunc->bGetMsgTxFlag();
        if (bMsgTxFlag)
        {
            if (CGlobalObj::GetIJ1939DIL()->DILIJ_SendJ1939Msg(pmCEexecuteFunc->dwGetNodeClientId(),
                    psTxMsg->m_sMsgProperties.m_byChannel,
                    psTxMsg->m_sMsgProperties.m_eType,
                    psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.unGetPGN(),
                    psTxMsg->m_pbyData,
                    psTxMsg->m_unDLC,
                    psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_uPGN.m_sPGN.m_byPriority,
                    ADDRESS_NULL,
                    psTxMsg->m_sMsgProperties.m_uExtendedID.m_s29BitId.m_uPGN.m_sPGN.m_byPDU_Specific) == S_OK)
            {
                Return = 1;
            }
        }
	}
	psTxMsg = 0;
	delete psTxMsg;
    return Return;
}

gt-morgott, 2015-03-18:
I detected that the problem occures on a 32-bit OS. 64-Bit OS don't have a problem with the construction of the data structure seemingly.

skylla, 2015-08-25:
When I recently started a nodesimulation project I got what I think is the same problem, the message is not transmitted correctly. But I am running 64 bit windows. The actual message that is transmitted varies, when sending different messages sometimes the data in m_pbyData is switched, sometimes it is transmitted on the wrong channel (m_byChannel) and other errors. I suspect that there is a problem with the STJ1939_MSG variable and that the send function gets some data from a memory position that belongs to another message.

Are there any more progress on this?