rbei-etas/busmaster #774: MHS hardware cound not be started

mibcat, 2015-02-27:
Hi !
with the following configuration I'm not able to connect to my Tiny CAN I-XL hardware

Windows 8.1 64 bit
BusMaster 2.6.1
Tiny CAN driver: Windows 8.1 V3.02

After selecting the driver and setup the channel configuration a connect leads to the trace window message: "Starten der Hardware fehlgeschlagen" -> hardware could not be started.

But the hardware works fine with the Tiny-CAN Monitor (a small demo program delivered with the MHS driver software).

Best regards,
Michael.

nostar, 2015-02-27:
In order for me to use the MHS TinyCAN with busmaster I have to select 'Channel Configuration' after selecting the MHS driver, then set the correct CAN baud rate and either clear the 'Tint-CAN serial number' input box or input the correct serial number.

mibcat, 2015-02-27:
Great - it works !! Thanks for that hint.
-> but the configuration dialog needs to be improved definitely ;-))

nostar, 2015-02-28:
Agreed, this is certainly a bug.

closed 2015-03-02 by RBEI-Saravanan