rbei-etas/busmaster #775: Busmaster 2.6.1 requires admin rights to run

eder-s, 2015-03-02:
Busmaster seems to require admin rights on some computers after installation. On my computer Busmaster needs admin rights to run. If I start it with a non admin account it freezes while loading the program(no crash, no additional information). Busmaster is installed to a folder which is writable without admin rights.
On a colleague's computer(same setup) everything runs fine without admin rights.
Both computer's run Win 7 64bit.
I have the same issues also with Busmaster 2.5

Does someone have a clue what the problem could be or faced a similar issue?

RBEI-Saravanan, 2015-03-06:
Busmaster needs admin rights to be launched. Could you please check the UAC level of the other computer in which it is used without any issues

eder-s, 2015-03-09:
UAC on the other computer is "standard", second option from top. My computer has the same setting.
Even if I set it to the lowest level Busmaster still crashs without admin rights. Is there any special windows privilege Busmasters needs which I could check?