rbei-etas/busmaster #781: Gigabox is mentioned inside FR Hardware but not selctable inside the setup.

RoSchCo, 2015-03-11:
Gigabox is mentioned inside FR Hardware but not selctable inside the setup. Also an error message is given if the Gigatronic driver is chosen.

RBEI-Prathiba, 2015-03-12:
Hello,

Gigatronik flex-i driver can only be used with latest drivers - v1.0.8 gigatronik flex-i driver library
and currently only CAN related Hardware Interfaces are listed inside the setup.

Regards,
Prathiba

RoSchCo, 2015-03-12:
Good morning!
Many thx for your update.
I have already installed the version 1.0.6. Unfortunately I am not able to see the 1.0.8 directly at Gigagronik. Where did you get the latest one?
Regards,
RS!

gt-morgott, 2015-03-16:
Hello,
as already mentioned the GIGABOX flex-i driver 1.0.8 is required to use the GIGABOX flex-i with BUSMASTER. You can find this driver library on the Gigatronik Homepage:
http://automotive.gigatronik.com/de/leistungen/service/gigabox-flex-i-software.html

Regards,
Michael

RoSchCo, 2015-03-16:
Hello,
I got the information but as already mentioned I was not able to download the driver 1.0.8. Only 1.0.6 was available.
Now with 1.0.8 I can configure the GIGABOX a. also I can have a look on the bus.
Many thx for the support.
Bye,
RS!