rbei-etas/busmaster #785: Make J1939 mode as functional as CAN mode

glafayette615, 2015-03-26:
It seems to me that most of the features available in CAN mode (i.e. 11 bit identifiers) are not available in J1939 mode.  For me, and I assume many others this is a severe limitation of this tool.

Am I missing something?  Is there a workaround method to send J1939 messages, log them into a graphical signal, etc?

Is there an idea of how big the undertaking would be to make the J1939 mode function the same way as the CAN?

RBEI-Prathiba, 2015-03-30:
Hi,
J1939 uses the 29-bit identifier defined in the CAN 2.0B protocol. To send J1939 messages please do the following:
1. Activate J1939 using J1939->Activate
2. To configure messages for transmission, use J1939 Transmit Window via menu 'J1939->Transmit'

Currently you can only log the data in to a file using menu 'J1939->Logging'.
For more details, please refer section 'J1939' in help file.

mthill, 2016-11-30:
Reiterating glafayette615's question, is there an idea of how big the undertaking would be to make the J1939 mode function the same way as the CAN?  

It would be a major improvement to be able to analyze J1939 data in the same manner as the CAN module.  We can do a signal window but not a signal graph.