rbei-etas/busmaster #786: Busmaster -- not able to open Function Editor

dinguluer, 2015-03-30:
Hi,

I am using Busmaster Version - 2.6.0.

I pressed the button LIN -->node simulator --> configure. Then add new node.
When i tried to add an empty C file to the node gets an error "open C file created for LIN".

Then i searched in help file & found that function editor is used to create LIN node files for bus master.

Is function editor similar to - CAPL Browser ?

I am not able to find how to open this Function editor to create new files ..?

Please suggest how to open function editor?

Regards,
Dinesh

RBEI-Prathiba, 2015-03-30:
Hi,
Please follow the below steps.
1. Select the node, Click on 'Add New File...' button, specify the file name (do not load empty file).
2. Click on 'Open' button, new file with the name specified will be created and opened in Function editor
3. Use function editor to define handlers
4. Once handlers are defined, click on 'Build and Load' and 'Enable All Handlers'

Note: To build the cpp/c files, please install MinGW component. Link to download MinGW and document with instructions is sent to your mail id. The same will be updated in the next version help file.

Regards,
Prathiba

dinguluer, 2015-03-31:
Thanks Prathiba, its working now.

Regards,
Dinesh

closed 2015-04-02 by RBEI-Saravanan