rbei-etas/busmaster #788: Unable to initialise handler functions for execution - After compilation

petar-penchev, 2015-04-23:
When I create a simulation with Busmaster 2.6.1 and if I add OnDLL_Load or OnDLL_Unload or OnKey_xxxx handler the simulation build successfully however the following error appears when Load button is pressed:
Unable to initialise handler functions for execution.

RBEI-Saravanan, 2015-04-24:
Could you check if the .def file is created in the same path of your .cpp file ?

petar-penchev, 2015-04-24:
Yes , The file is there. Here it is the contents.

EXPORTS
vSetEnableLoggingProcAddress
vSetDisableLoggingProcAddress
vSetWriteToLogFileProcAddress
vSetConnectProcAddress
vSetDisconnectProcAddress
vSetGoOnlineProcAddress
vSetGoOfflineProcAddress
vSetStartTimerProcAddress
vSetStopTimerProcAddress
vSetSetTimerValProcAddress
vSetEnableMsgHandlersProcAddress
vSetEnableErrorHandlersProcAddress
vSetEnableKeyHandlersProcAddress
vSetEnableDisableMsgTxProcAddress
vSetSendMsgProcAddress
vSetGetDllHandleProcAddress
vSetTraceProcAddress
vSetProcAddress
bGetProgramVersion
vSetKeyPressed
vSetGetMessageName
vSetTimeNow
vSetGetFirstCANdbName

OnMsgName_WRITECMD
OnMsgName_RESPONSE_REUQEST
OnKey_d
OnDLL_Load
OnDLL_Unload

.DLL is compiled properly, however it gives me 0 Handlers when I build it.

petar-penchev, 2015-04-24:
And this is the MinGW version I am using

gcc-v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=c:/mingw/bin/../libexec/gcc/mingw32/4.8.1/lto-wrapper.exe
Target: mingw32
Configured with: ../gcc-4.8.1/configure --prefix=/mingw --host=mingw32 --build=m
ingw32 --without-pic --enable-shared --enable-static --with-gnu-ld --enable-lto
--enable-libssp --disable-multilib --enable-languages=c,c++,fortran,objc,obj-c++
,ada --disable-sjlj-exceptions --with-dwarf2 --disable-win32-registry --enable-l
ibstdcxx-debug --enable-version-specific-runtime-libs --with-gmp=/usr/src/pkg/gm
p-5.1.2-1-mingw32-src/bld --with-mpc=/usr/src/pkg/mpc-1.0.1-1-mingw32-src/bld --
with-mpfr= --with-system-zlib --with-gnu-as --enable-decimal-float=yes --enable-
libgomp --enable-threads --with-libiconv-prefix=/mingw32 --with-libintl-prefix=/
mingw --disable-bootstrap LDFLAGS=-s CFLAGS=-D_USE_32BIT_TIME_T
Thread model: win32
gcc version 4.8.1 (GCC)

RBEI-Prathiba, 2015-04-29:
Hi,
Thanks for the details. Could you please send the .dll file to busmaster@in.bosch.com mail id for further analysis.