rbei-etas/busmaster #798: Spurious CAN messages displayed

avogra, 2015-05-27:
Short description:
I receive a lot of spurious / unreasonable can-messages when starting to send CAN messages from Busmaster (see screenshot). It looks like the internal bus error counter would be displayed in the can message window.

Environment:
Version is 2.6.2, Operating system is Windows 7 x64 german, CAN-Interface is Vector VN1630A with CANpiggy 1054mag and Vector driver version 8.9.22.

Detailed description:
My DUT is initially in sleep mode and to be woken by CAN. I load the configuration and setup CAN interface and channel. I then start the configuration with "Start stop sending messages" disabled. Then I enable sending messages. After some errorframes (due to startuptime of my dut) communication is working fine.
But for every CAN message sent (or tried) by the Busmaster, i get displayed one CAN message with unreasonable ID. It starts at ID 0x808 and increases in steps of 0x800 up to >= 0x7F00. As soon as communication is error free, i still get a spurious message for each msg sent but with IDs decreasing in steps of 0x100. When the ID reaches 0x0, no more spurious messages are displayed
[image]

RBEI-Prathiba, 2015-05-28:
Bus Error occurs only if there is a mismatch in baud rate or Channel configuration parameters. Does these errors occur only if the device is in sleep mode initially and woken by CAN later? What happens if the device is already in wake up mode does it still receive errors?

avogra, 2015-05-28:
When the device is awake and i start sending messages from Busmaster, there are no bus errors and no extra messages.

I expect to get some bus errors until the the device is awake as there is noone else setting the ACK-bit. What I am surprised about are those messages in between the errors and after each TX message.

RBEI-Prathiba, 2015-05-29:
Messages between errors and each Tx message should be transmitted by the device/ECU in response to the Tx message. Tool displays all the message received on the CAN bus. As we don't have the setup, could you please check if the these messages are transmitted by device.

avogra, 2015-05-29:
These messages are not transmitted by the device. It is not able to send extended messages (IDs go up to 7Fxx). When I use the same setup with Vector Canoe, those messages are not visible.

Just a guess from my side: Is it possible, that the Vector driver transmits information about the bus error state and Busmaster is interpreting this information as CAN messages? the sequence of "IDs" would fit, if the high byte is the error counter: increase by 4 for every unsuccessful message, decrease by 1 for every successful message.

avogra, 2015-06-11:
I just had the glorious idea to record a log ;-) This should help to understand my point. I can only attach images, so i paste the content at the end of this post. What you will see:
In the beginning the Busmaster tries to send message ID 0x10 but the device is in sleep and sends no ack. therefore only errorframes (0x2001) are detected. at the same time, spurious messages (0x6A04, 0x7204, 0x7A04, ...) are displayed, which stop at 0x8202. When the device starts to set the ack bit, busmaster sucessfully sends message 0x10. for each message sent, a spurious message with decreasing ID (0x8102, 0x8002, 0x7F02,...) which stop when the ID would be < 0. The ocassional ID 0x150 is actually sent by the device. This is the LOG from Busmaster:
```
***BUSMASTER Ver 2.6.2***
***PROTOCOL CAN***
***NOTE: PLEASE DO NOT EDIT THIS DOCUMENT***
***[START LOGGING SESSION]***
***START DATE AND TIME 11:6:2015 9:58:28:120***
***HEX***
***SYSTEM MODE***
***START CHANNEL BAUD RATE***
***CHANNEL 1 - Vector - VN1630A Channel 2, Seriennummer- 500496 - 100000 bps***
***END CHANNEL BAUD RATE***
***START DATABASE FILES (DBF/DBC)***
***C:\Sandbox\rr_konzept_sh\40_Implementierung\20_SW_Entwicklung\30_Restbus_Simulation\MY16_NM\sternenhimmel_controlbus.dbf***
***END OF DATABASE FILES (DBF/DBC)***
***<Time><Tx/Rx><Channel><CAN ID><Type><DLC><DataBytes>***
09:58:32:0701 Rx 1 0x6A04 s 0 
09:58:32:0701 Rx 1 0x2001 x 0 
09:58:32:0707 Rx 1 0x7204 s 0 
09:58:32:0707 Rx 1 0x2001 x 0 
09:58:32:0714 Rx 1 0x7A04 s 0 
09:58:32:0714 Rx 1 0x2001 x 0 
09:58:32:0721 Rx 1 0x8202 s 0 
09:58:32:0721 Rx 1 0x2001 x 0 
09:58:32:0729 Rx 1 0x2001 x 0 
09:58:32:0736 Rx 1 0x2001 x 0 
09:58:32:0744 Rx 1 0x2001 x 0 
09:58:32:0751 Rx 1 0x2001 x 0 
09:58:32:0758 Rx 1 0x2001 x 0 
09:58:32:0766 Rx 1 0x2001 x 0 
09:58:32:0773 Rx 1 0x2001 x 0 
09:58:32:0781 Rx 1 0x2001 x 0 
09:58:32:0788 Rx 1 0x2001 x 0 
09:58:32:0795 Rx 1 0x2001 x 0 
09:58:32:0803 Rx 1 0x2001 x 0 
09:58:32:0810 Rx 1 0x2001 x 0 
09:58:32:0818 Rx 1 0x2001 x 0 
09:58:32:0825 Rx 1 0x2001 x 0 
09:58:32:0832 Rx 1 0x2001 x 0 
09:58:32:0840 Rx 1 0x2001 x 0 
09:58:32:0847 Rx 1 0x2001 x 0 
09:58:32:0855 Rx 1 0x2001 x 0 
09:58:32:0862 Rx 1 0x2001 x 0 
09:58:32:0869 Rx 1 0x2001 x 0 
09:58:32:0877 Rx 1 0x2001 x 0 
09:58:32:0884 Rx 1 0x2001 x 0 
09:58:32:0892 Rx 1 0x2001 x 0 
09:58:32:0899 Rx 1 0x2001 x 0 
09:58:32:0906 Rx 1 0x2001 x 0 
09:58:32:0914 Rx 1 0x2001 x 0 
09:58:32:0921 Rx 1 0x2001 x 0 
09:58:32:0929 Rx 1 0x2001 x 0 
09:58:32:0936 Rx 1 0x8102 s 0 
09:58:32:0936 Tx 1 0x010 s 1 00 
09:58:32:1201 Rx 1 0x8002 s 0 
09:58:32:1201 Tx 1 0x010 s 1 00 
09:58:32:1699 Rx 1 0x7F02 s 0 
09:58:32:1699 Tx 1 0x010 s 1 00 
09:58:32:2200 Rx 1 0x7E04 s 0 
09:58:32:2200 Tx 1 0x010 s 1 00 
09:58:32:2701 Rx 1 0x7D04 s 0 
09:58:32:2701 Tx 1 0x010 s 1 00 
09:58:32:3201 Rx 1 0x7C04 s 0 
09:58:32:3201 Tx 1 0x010 s 1 00 
09:58:32:3701 Rx 1 0x7B04 s 0 
09:58:32:3701 Tx 1 0x010 s 1 00 
09:58:32:4200 Rx 1 0x7A04 s 0 
09:58:32:4200 Tx 1 0x010 s 1 00 
09:58:32:4701 Rx 1 0x7904 s 0 
[... 259 lines cut ...]
09:58:39:0703 Tx 1 0x010 s 1 00 
09:58:39:0944 Rx 1 0x150 s 8 53 53 53 00 00 00 00 00 
09:58:39:1203 Tx 1 0x010 s 1 00 
09:58:39:1703 Tx 1 0x010 s 1 00 
09:58:39:2203 Tx 1 0x010 s 1 00 
09:58:39:2703 Tx 1 0x010 s 1 00 
09:58:39:3204 Tx 1 0x010 s 1 00 
09:58:39:3703 Tx 1 0x010 s 1 00 
09:58:39:4204 Tx 1 0x010 s 1 00 
...
...
...
09:58:47:2706 Tx 1 0x010 s 1 00 
09:58:47:3207 Tx 1 0x010 s 1 00 
09:58:47:3706 Tx 1 0x010 s 1 00 
09:58:47:4207 Tx 1 0x010 s 1 00 
09:58:47:4707 Tx 1 0x010 s 1 00 
09:58:47:5206 Tx 1 0x010 s 1 00 
09:58:47:5706 Tx 1 0x010 s 1 00 
09:58:47:6207 Tx 1 0x010 s 1 00 
09:58:48:0943 Rx 1 0x150 s 8 53 53 54 00 00 00 00 00 
09:58:49:0943 Rx 1 0x150 s 8 53 53 54 00 00 00 00 00 
***END DATE AND TIME 11:6:2015 9:58:50:176***
***[STOP LOGGING SESSION]***
```

RBEI-Prathiba, 2015-07-30:
Messages with ID (0x6A04, 0x7204, 0x7A04, ...) are the error messages received from the hardware. BusMaster follows convention for error ids as (0x6A04, 0x7204, 0x7A04, ...), hence the error message with the same id will not be seen in CANOe.

RBEI-Srinibas, 2015-12-08:
Error ids should not be displayed as like actual messages (Tx/Rx).

almadun, 2016-12-01:
I have question regarding the <DataBytes> I have can data of GYROSCOP
how can I convert the raw data to physical data in order to get the three axises data (X,Y,Z) , the data I have like in below and what I need to convert is the DataBytes data 

"***<Time><Tx/Rx><Channel><CAN ID><Type><DLC><DataBytes>*******"
0:0:0:0000 Rx 1 0x220 s 8 41 04 11 04 BC 00 05 10