rbei-etas/busmaster #799: Busmaster do not find .cpp file

sandy67, 2015-05-29:
Hello,
I have a problem with the bus master of Github version 2.6.2 under
Windows 7.

When installing the .exe told me, the  MinGW Compiller
already exists.
The PATH environment variable C: \ MinGW \ bin is correctly set.

Measuring on the CAN bus works without problems.

Now I want to insert a node, so I open  my window
Simulated system. There I joined a new simulated system and a
new node.
 
Then i added an existing .cpp file witch already worked with an older Version of busmaster under Windows XP.

Then I want to compile the new node, and press the Build
Button.

Now the problem appears:

Although my Progrmm node.cpp under
C: \ User \ bus master \ SimulatedSystem \ Win7Test \ node1.cpp 
exists, the busmaster tells me: 

 The file is not properly build

In Output Window you can see:
Invoking MinGW gcc compiler Failed: The system can not found the specified file.

How do I get the bus master to find the file ?

RBEI-Prathiba, 2015-05-29:
Could you please create new cpp file and check if it is getting compiled using MinGW? This is to confirm if the issue is with invoking MinGW or building old cpp files.

sandy67, 2015-05-29:
I already tried it with an new file, with the same result: The Busmaster told me, that he do not find the file

RBEI-Prathiba, 2015-05-29:
Looks like issue with MinGW, could you please uninstall or rename the old MinGW  and install the new one. Please refer help section 'CAN->MinGW Installation using TDM-GCC Installer' for details

RBEI-Prathiba, 2015-05-29:
Before uninstalling could please execute 'mingw32-make -v' in command prompt and let us know what version of MinGW you have.

sandy67, 2015-06-01:
in which folder do I have to go to execute this command?

RBEI-Prathiba, 2015-06-01:
You can execute this from command prompt from any folder as environment variable is set in path variable

sandy67, 2015-06-01:
I executed it from C:\MinGw\ and from C:\MinGw\bin but the cmd always tells me: The command "mingw32-make -v" is either falls written or could not be find

sandy67, 2015-06-01:
sorry: false written not falls written

sandy67, 2015-06-01:
I got it:

c:\MinGW\bin>make -version
GNU Make 3.82
Built for i686-pc-mingw32
This program is built by Equation Solution <http://ww
for Windows.
Copyright (C) 2010  Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gn
This is free software: you are free to change and red
There is NO WARRANTY, to the extent permitted by law.

Is this the correct Version vor the Busmaster V 2.6.2?

RBEI-Prathiba, 2015-06-01:
BusMaster uses MinGW version of GCC, could you please uninstall or rename the old MinGW and install the new one. Please refer help section 'CAN->MinGW Installation using TDM-GCC Installer' for details

sandy67, 2015-06-01:
ok, I installed the newest MinGW Version, now it works, the Busmaster is able to find the file and compile it.
Thank you for your help
kind regarts,
sandy

closed 2015-06-01 by sandy67

ketanrathod90, 2020-11-11:
> BusMaster uses MinGW version of GCC, could you please uninstall or rename the old MinGW and install the new one. Please refer help section 'CAN->MinGW Installation using TDM-GCC Installer' for details

I did the same but I am getting following error: Invoking MinGW gcc compiler Failed: The system cannot find the file specified.