rbei-etas/busmaster #802: Associate Database to specifik BUS

MattiasRosengren, 2015-06-03:
Hi,

Is there a way to tie a "Database association" to a specific can-bus while listening to more than one bus?
The same CAN-id could have a completly diffrent meaning on the other bus.

regards
Mattias Rosengren
Koenigsegg

nostar, 2015-06-03:
No.  This is a huge need for me, and it was promised for 3.0.0 on the Rease Roadmap page but now its gone.

AutologicAU, 2015-10-19:
I also think this is a good idea, perhaps this can be defined channel wise on opening or associating the database.
Also it would be nice to have master database with extended properties that we can export to a to a sub or child database so that we do not need to retype messages in new database files.

Teddyz, 2018-07-05:
Yes please. This is also a problem for me.