rbei-etas/busmaster #804: no filter on std:can-fd frames  (busMaster 2.6.1)

klausmueller1, 2015-06-15:
I tried to setup a project with can-fd (ETAS ES523) and want to use filter on the receive frames, but the filter didn't work. I think the problem is related to the frame types : std:can-FD  and  ext:can-FD !?

RBEI-Prathiba, 2015-06-15:
CANFD support is not available on all features, currently it is only added only for Node Simulation, Transmission (limited to 8 bytes) and Reception.

klausmueller1, 2015-06-15:
Hello,
Okay, so you see this as a new requirement ?  Is there a planning to support all features of CAN also on CAN-FD ?
Or at least a planning - which missing feature on CAN-FD shall be implemented in one of the following versions on CAN-FD ?

Kind Regards
Klaus Mueller