rbei-etas/busmaster #805: Busmaster 2.5.0 & 2.6.2

Phreak-Show, 2015-06-15:
When trying to connect to a Vector VN1630 (newest drivers), Busmaster 2.5.0 and 2.6.2 say that the hardware can not be started. I added busmaster to the CAN channels in Vector Hardware config, set all channels to 500kBit and click connect.
Then the message appears in trace window.

OS is Windows 8.1 x64, and it also happens with Vector CANCaseXL.

RBEI-Prathiba, 2015-06-15:
It could be with Vector XL driver library, could you please try installing Vector XL driver library for Windows 8.

Phreak-Show, 2015-06-15:
Newest Windows 8 lib is installed. Other programs (CANoe) work fine with the hard- and software configuration. I did not try to tun them simultaneously.