rbei-etas/busmaster #809: Pannel support Busmaster

dinguluer, 2015-06-25:
Hi,

Do busmaster supports for Pannel build ?

I saw this link :---
https://github.com/rbei-etas/busmaster/issues/84

If not, can i create a pannel in QT creator & send/receive CAN bus event using busmaster server ?

If it is possible, can you guide me to the documentation to access busmaster server ?

Regards,
Dinesh

RBEI-Saravanan, 2015-06-29:
Thank you for the interest.
Panels is not yet supported. Yes, that a good idea to use QT creator and use BusMaster server for send and receive.

Please refer to the help file on the below topics for APIs.
CAN ==> COM Interface

closed 2015-08-19 by RBEI-Saravanan

RBEI-busmaster, 2017-07-19:
Hello,
BusMaster supports Instruments add-on to create GUI and use it for communication. 
Please refer http://rbei-etas.github.io/busmaster/Add_Ons.html for more details

yijoung, 2017-08-16:
So it means that Instruments add-on is not free and it's paid?