rbei-etas/busmaster #812: undefined reference to 'SendMsg'

e1806, 2015-07-09:
Hi,

I tried to build the CAN-example from the busmaster user manual.
My node simulation build will not properly build. 
This my output:

In file included from D:\Tests\busmaster\cpp_test.cpp:12:0:
D:\Tests\busmaster\Brose_komplett_Unions.h:5:18: warning: malformed '#pragma pack(pop[, id])' - ignored [-Wpragmas]
 #pragma pack(pop,8)
                  ^
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0x8c): undefined reference to `SendMsg'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0xa6): undefined reference to `Connect'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0xc1): undefined reference to `Trace'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0xd5): undefined reference to `EnableLogging'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0xe9): undefined reference to `DisableLogging'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0x192): undefined reference to `SendMsg'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0x1b0): undefined reference to `Trace'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text+0x1cb): undefined reference to `Trace'
D:\Tests\busmaster\cpp_test.o:cpp_test.cpp:(.text$_ZN18_InternalStructCANC1Ev[_ZN18_InternalStructCANC1Ev]+0x14): undefined reference to `Trace'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [DllFile.dll] Error 1
D:\Tests\BUSMAS~1\DLLMake:4: recipe for target 'DllFile.dll' failed

I don't know why, because the functions are Busmaster API functions?
I checked the Path,... and searching here but i don't find a solution.

Can someone help me?

My environment:
Win7, Busmaster 2.6.2, tdm64-gcc-5.1.0-2.exe

RBEI-Prathiba, 2015-07-10:
Hi,
From the environment details, we understand that MinGW 64bit is installed. Can you please try installing MinGW 32 bit. For more details please refer help Section 'CAN->MinGW Installation using TDM-GCC Installer'

e1806, 2015-07-10:
Hi,

I tried the 32-bit version and it seems to be ok. No Errors by compiling the example.

Thanks for now

closed 2015-07-13 by RBEI-Saravanan