rbei-etas/busmaster #813: Support for new ES581.4

psyklopz, 2015-07-09:
ETAS no longer sells the ES581.3 and has replaced it with the ES581.4.  Busmaster doesn't seem to support this adapter in any way.  Please add support for this new adapter.

RBEI-Prathiba, 2015-07-10:
ES581.4 can be accessed using 'CAN->Driver Selection->ETAS BOA', if you have Boa framework (v2.0 and above) installed.

closed 2015-08-04 by RBEI-Saravanan

Pdeokar, 2015-11-09:
Having issues with Driver Selection when using ES581.4 and Busmaster 2.6.3. It gives 'Driver Selection Failed". Working on Windows 7. 
I installed USB driver for ES581.4 through HSP (V10.10.0) and also upgraded firmware on to V3.0.0. But when try to select driver => CAN\Driver Selection\ETAS BOA from the main menu, it gives me 'Driver Selection Failed' .

I also tried changing BusMaster compatibility mode to Windows XP too, didn't help.

psyklopz, 2015-11-09:
@Pdeokar You need to install the BOA Framework, available here:
http://www.etas.com/en/downloadcenter/18102.php

Make sure to restart afterwards.
It's not included on the CD that comes with the ES581.4.

Pdeokar, 2015-11-09:
It worked! Thanks

ranjandhingra923, 2016-02-09:
i have installed BOA framework. i can successfully select DRIVER as ETAS BOA, can further assign CAN1 to channel 1, but after that when i connect , NO signals are shown in the CAN window. Can you assist as what can be the cause ?