rbei-etas/busmaster #817: Not able to run busmaster using Value can3 in my pc, attaching the error snapshot.

rpamparthi, 2015-07-23:
[image]

RBEI-Prathiba, 2015-07-24:
Please provide us few more details:
1. Which version of BusMaster is used
2. Steps to reproduce the same

rpamparthi, 2015-07-24:
1. I am using 2.6.2 version of busmaster.
2. I have installed the value can3 drivers, when trying to select the device driver for value can in busmater i am getting that error.
3. I tried reinstalling the device drivers for value can3 and busmaster software but seeing the same error.

RBEI-Prathiba, 2015-08-03:
As we don't have the hardware setup with us, we would like to have a WebEx to analyse the issue. Please send your contact details to BUSMASTER@in.bosch.com.

ian2211, 2015-08-11:
I am currently receiving the same error message.
[image]

My vcand32.dll in \Windows\System32 folder is version 3.5.1.41  and the firmware on my ValueCAN is 1.7. I am using Windows 7 and Busmaster version 2.6.3

This is a new laptop I am using, but busmaster works on my old laptop with the same hardware and CAN driver configuration. I have tried going back to Busmaster 2.6.2 but the error is the same

RBEI-Prathiba, 2015-08-13:
Please check if all the drivers installed in new laptop are same as the old laptop.If new laptop configuration is different from old laptop configuration (OS) then download respective drivers.

houghtonbrad, 2015-11-24:
Has there been any success with this issue? I'm having the exact same problem. Only with my issue I'm using the same computer but a different ValuCAN device and reinstalled the drivers that came with it.

PavanDendi, 2016-01-21:
I am having the exact same issue as the others mentioned here.  The problem happens on two machines, both Windows 7.  Here is the contents of the error message that Windows gives me:

Problem signature:
  Problem Event Name:	BEX
  Application Name:	BUSMASTER.exe
  Application Version:	2.0.0.0
  Application Timestamp:	564ad434
  Fault Module Name:	CAN_ICS_neoVI.dll
  Fault Module Version:	1.0.0.1
  Fault Module Timestamp:	564ad254
  Exception Offset:	00005ce1
  Exception Code:	c0000409
  Exception Data:	00000000
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	77af
  Additional Information 2:	77afe6d357f70292e0a94fcd7d04ce3d
  Additional Information 3:	9163
  Additional Information 4:	91638fcf069f89e86a7df38007c60a04

BEX and Exception Code c0000409 seem to indicate a Data Execution Prevention problem, but Windows refuses to let me turn off DEP for Busmaster.exe.  When trying to add it to the exception list, Windows gives the following error message:

This program must run with data exception protection (DEP) enabled.  You cannot turn off DEP for this program.

I have also tried disabling DEP completely through the command prompt and the BIOS, but the problem still persists with the same error message.

I am using Busmaster 2.6.4

houghtonbrad, 2016-01-21:
I had exchanged emails back and for with Prathiba and it came down to it being a hardware issue. This being the case they were not able to find the exact cause. I hope you have Breyer luck with your situation.

PavanDendi, 2016-01-25:
@houghtonbrad
Looks like they've already patched it.  I came across the file by sheer dumb luck as I was poking around the repository.  It fixed the issue for me; hopefully it will help you out as well.

You can find the patch here: https://github.com/rbei-etas/busmaster-transient