rbei-etas/busmaster #818: Issue with ELM327

arildojr, 2015-07-23:
Hi,

I´m using here BUSMASTER software. I´m trying to send CAN data through a ES581 driver to a ELM327 OBD 2 Bluetooth Scanner. Although, I always receive “Bus Error” in message window when sending data. I basically use the USB ES581 (120 ohms resistor included) connected to the OBD port of ELM327 through CAN High and CAN Low pins.
There is any issue of BUSMASTER related to ELM327 OBD 2? Do you guys know any specific configuration related to ELM327 OBD 2 in BUSMASTER?

Best regards,

RBEI-Prathiba, 2015-07-24:
Please let us the know hardware used (ES581.2 or ES581.3).
There is no such specific configuration required in BusMaster related to OBD 2, as BusMaster communicates with ES581 and it depends on the connectivity from ES581 to OBD port.
To confirm the same, you could perform a test by connecting ES581 channels in loop back (with 120 ohm resistor) and check if you still receive error messages on transmission.

arildojr, 2015-07-31:
ES581.3
Channels in loop back fixed my issue.
Thank´s

closed 2015-07-31 by arildojr