rbei-etas/busmaster #820: Bus Error when baud rate is set to 500kbps

deepakomanna, 2015-07-29:
Hi !
When I set the busmaster baud rate to 250kbps then the data is transmitting and no bus error. But when I set the baud rate at 500kbps in busmaster & in software then I am getting the bus error. See attached files:CAN_Bus_Error.PNG & NM.cfx
[image]

Please suggest to get resolve out of this error

RBEI-Prathiba, 2015-07-30:
Error messages are received, If there is mismatch in baud rate between the controller and ECU. To confirm the same you can try connecting 2 channels in loop back with the same baud rate.

deepakomanna, 2015-07-30:
Hi !
The baud rate between controller and ECU is same 500kbps.
As you can see in the attached image file & NM.cfx file that Busmaster Baud rate set to 500kbps. 
Also we have configured the ECU to 500kbps.

arildojr, 2015-07-31:
Connect 2 channels in loop back, this solved my problem here.

deepakomanna, 2015-08-03:
HI !
I connected Channel 2 in look Back at Baud Rate = 500kbps. I got the bus error (See attached file- Channel 2 Loop Back). But if I set the Baud rate = 250kbps there is no error.
I read that value can 3 is supports upto 1mbps. 
So not able to understand where is the problem i.e. Value Can 3 or Busmaster setting.
[image]

RBEI-Prathiba, 2015-08-03:
Could you please try connecting to 120 ohms resistor