rbei-etas/busmaster #825: BUSMASTER 2.6.3 can diagnostic don't transmit flowcontrol

chenqianglight, 2015-08-10:
0	00:02:02.280.0	0x00000768		0x08	     03 22 F1 81 00 00 00 00 	
1	00:02:02.290.0	0x00000708		0x08	     03 7F 22 78 FF FF FF FF 	
2	00:02:02.290.0	0x00000708		0x08	     10 10 62 F1 81 01 05 4E 	
768 busmaster transmit
708 ecu transmit
the number 1 frame means that needs to extended p2cansever time,
after the ecu transmit the fram(number 2),busmaster don't transmit flowcontrol

sd886189, 2015-08-14:
I had the same problem!!  UP UP

bentearadu, 2015-08-19:
Casian fixed the issue.

Saravanan, he will send you an e-mail with what modification need to be done (FYI - only one line should be deleted)

RBEI-Saravanan, 2015-08-20:
That sounds great. Looking forward for the details.

mantoni86, 2015-09-28:
Hello, please post a solution or a new Version, i also have this problem and need a solution

bentearadu, 2015-09-29:
I think this was already fixed in the new SW.
Ask Saravanan about this.

mantoni86, 2015-09-29:
So how do I write him? Can he send me an eMail? Or is it possible to write him a Message?

RBEI-Prathiba, 2015-09-30:
We have received the fix for the flow control issue and the same will be part of the next release.

RBEI-Prathiba, 2015-11-17:
Please check with v2.6.4

mantoni86, 2015-11-20:
Hello, now it works, thank you.

But with Blocksize 0 there is still a problem. But when I set Blocksize to != 0, it works good.

Mit freundlichen Grüßen / Best regards

Martin Antoni
AE/EEG1

Tel. +49(711)811-12468

GuillaumeHM, 2017-03-29:
Hello all,
I still have this issue on version 3.1. It was already there in 3.0 if I remember right.
When I use the Main Diagnostic Window, I have quite the same log as in the first comment.
My workaround consist to manually throw a flow control frame thanks to Transmit Window.
Am I the only one ?
BR
Guillaume