rbei-etas/busmaster #828: FlexRay and ES583.1

sandy67, 2015-08-18:
Hi,
I am working with Busmaster 2.6.2 and I have two questions:

1. Is it possible to start the FlexRay Bus with the Busmaster? If yes, how can I do this?

2. Why is it not possible to select 2 FlexRay  channels, although on ES583.1 two channels are available?

Many Thanks,
Sandy

sandy67, 2015-08-18:
Hi, two more questions:
3. Why ist it not possible to ad a new frame in the Configure Transmission Messages Window?
4. How can I log the traffic on the FlexRay and replay it?

ETAS-Nithin, 2015-08-24:
Hi Sandy,

Please find the response to your queries 

1. Is it possible to start the FlexRay Bus with the Busmaster? If yes, how can I do this?
- Current version does not support this feature. This feature will be implemented as part of an extended add-on for FlexRay in the future

2. Why is it not possible to select 2 FlexRay channels, although on ES583.1 two channels are available?
- The ES583.1 has single FlexRay node with channels A & B. For the moment Busmaster only supports A & B on single FlexRay node. We shall test this once again and revert to you.

3. Why ist it not possible to ad a new frame in the Configure Transmission Messages Window?
- ECUs from the Cluster Configuration Window to be selected first. The frames defined for the selected ECUs would be available in the Tx window to be added.

4. How can I log the traffic on the FlexRay and replay it?
- Current version does not support this feature. This feature will be implemented as part of the extended add-on as mentioned in response to Query 1.

We shall keep you updated once the above features are made available in the extended add-on.

Regards
Nithin

sandy67, 2015-08-24:
Hi Nithin,
thanks a lot for yourr response.
Yes, please send me an update note when the features are made available.

Regards,
Sandy

ETAS-Nithin, 2015-08-24:
Sure Sandy

srikanthaddala, 2015-12-17:
HI is flexray supported currently, for callibration and other support from BUSMASTER