rbei-etas/busmaster #829: LIN Configuration of Busmaster (v2.6.2)

hfabian, 2015-08-19:
Hello,

I have a problem with the configuration of Busmaster v 2.6.2.

I want to communicate via LIN. I read the ldf in "Cluster Configuration".
I´ve selected the Schedule in "Schedule Configuration".
I have connected Busmaster with the Hardware. Therefore I've used PEAK USB Pro FD.

I have to send the master frame. This I've done in the "Configure transmission Messages" window.
Now I want to send the Header of the slave frame (Header only).
I don´t need the Slave message, this will come from my ECU.
I need a way to send only the Header of the message.

Thanks for you help.

hfabian, 2015-08-19:
I also con not edit the master frame.
When I disconnect from hardware, I can only edit the databytes in the slave mode. 
In my case the master frame sends the data for a LED display. So I have to read the data from the ecu in the slave datas to set the value of the display.
How can I set the data of the master frame (online)?
The data from master and slave are not the same, so I have to implement a little program.´
What should I do for this issue?

RBEI-Prathiba, 2015-08-20:
To send master request (header frame) from 'Transmit Window', select 'Type' as 'Master' and enable master mode in 'LIN->Cluster Configuration' using 'Enable Master Mode' check box.
To send Slave response from 'Transmit Window', select 'Type' as 'Slave' and configure data bytes.
For more details, please refer 'LIN->Transmit Messages'

hfabian, 2015-08-21:
That is what I have done.

I have the problem that my ecu doesn´t answer.
There is no Slave Frame incoming.

The display is not working with constant brightness.
The configuration of the display brightness use the Master Frame. There is a value range from 0 to 255 for the display brightness. 

Can the values of the frame be set direct? Not with the configure transmission window. Here I can not see the values, I see only the Message Data.
Can I see the values in Signal Data, like in case of CAN?

hfabian, 2015-08-21:
[image]

Why are the data from Configure transmission tessages window not the same in the messages window?
Why set Busmaster the values in byte 5, 6 and 7 to 00?

hfabian, 2015-08-21:
new version, new bugs.

I´m working now with the newest release V2.6.3
Now I have communication with the ecu.
Problem is now that the Message Window - LIN doesn´t show the messages. The display of the ecu show the value of the code of the master frame. The brightness is also in the master frame encoded, but it´s constant. I can not set the brightness to values from 0 to 255. So is the brightness in the master frame encoded.

I can set the value in the master frame but my display shows the same brightness all the time.

In the new version the values can be set like in CAN, that's good :-)