rbei-etas/busmaster #830: Send Messages with LIN

hfabian, 2015-08-24:
hello,
i have a problem sending messages in LIN.

I want to send with Busmaster for simulating the master and get the slave data from a real ecu.

In the "Configure Transmission Messages - LIN" window, I set the Data Bytes.
[image]

but in the Message Window -LIN I have other Data

[image]

Why set Busmaster the Values of the Bytes 5, 6 and 7 to zero?
How can I set this Bytes to the Values i want?

Thanks

RBEI-Prathiba, 2015-08-27:
Please let us know the below details:
1. Which hardware used
2. Who is sending Master frame, ECU or BusMaster?
3. What is the dlc expected by Master frame?

hfabian, 2015-08-31:
Hardware is PCAN-USB Pro FD.
BusMaster send the Master Frame, the ECU is the Slave.

I want to send the Data Bytes 46 EB 27 F5 EB 41 FC but BusMaster sends the Data Bytes 
46 EB 27 F5 00 00 00.

I have the same problem with a Vector CANcase.