rbei-etas/busmaster #834: No signal watch window

nostar, 2015-09-01:
No window appears when I click the icon or click Activate from the Signal Watch menu.  In all test cases I have at least 2 signals added to the signal watch list.  Tested on 2.6.2 and 2.6.3

RBEI-Prathiba, 2015-09-15:
We are unable to reproduce the same.Please provide the steps/scenario for further analysis.

nostar, 2015-09-16:
Now confirmed on both CAN and LIN.  I simply open any simulation, add some signals to the watch list for CAN and/or LIN, and activate the window, and no window appears.  Here is a screenshot where the show/hide signal watch window buttons are clearly pressed for both CAN and LIN.  Then I did a Window->Cascade, and there are no signal watch windows.
[image]

RBEI-Prathiba, 2015-09-24:
Hi,
Please send the .cfx file used to BUSMASTER@in.bosch.com for further analysis