rbei-etas/busmaster #836: load a pre compiled node simulation..

mgiaco, 2015-09-04:
Hello,
Is there a way to load a pre compiled node simulation dll ? I want to share the *.dll and *.o with other users.

thanks 
mathias

electrojit, 2015-09-06:
Yes it is possible.

In the simulation setup - right click on the node -> Edit Node ->Browse for dll 

Note: Make sure the dll file name should not be duplicate of any other node.

[image]

mgiaco, 2015-09-07:
thanks works perfect!

mgiaco, 2016-01-21:
Does not work with the new Version. That´s really bad.

Please can you fix that.

Thx

RBEI-Srinibas, 2016-01-22:
Hi,
Could you please try the below steps in new version of Busmaster -2.6.4 .

1. Open the Node simulation Window.
2. Right click the Root Node(<Protocol> Bus) and select "Add".
3. Then "Node Details" dialog box will appear. Add unique Node name and add existing .cpp/.dll .
Press OK.

Thanks,
Srinibas

mgiaco, 2016-01-22:
Hello,

So what you described is the normal way to add a Node Simulation - That´s not the problem I already know that.

The problem is that I want to load a precompiled Node Simulation File on a College´s PC. The DLL he used is compiled by me. So my college hasn´t installed the MinGW Compiler.

Is there a way to load such a precompiled DLL without compilation? It works in Version 2.6.3 as I remember.  

thx

RBEI-Srinibas, 2016-01-22:
Please share the error that you are getting in trace Window .

githublonelyone, 2018-11-09:
hello ,
i don't know how to load a DLL file in node simulation,anybody can help me?
l Add a dll file in CAN bus dialog then add a event void onDLL_load(),but no code in this function.but l can't use the api of dll file,because lt's wrong when debug.