rbei-etas/busmaster #840: Logging inside FR Cluster

RoSchCo, 2015-09-16:
Hello!
Is it planned to implement Logging for FlexRay? If yes do you know when?
Many thx in advance.

ETAS-Nithin, 2015-10-15:
Hello,
Apologize for the delay in response. 
Logging and further advanced features are planned for FlexRay in 2016. In principle development should be completed by Q2'16.

RoSchCo, 2015-10-15:
Hello Nithin!

Ok, many thx for the information. 

One additional question (I hope it is possible to do it via this way): 
We have one issue. We have stable FR communication with our ECU. If we 
have a planned reset on the ECU (so also of the FR communication) the 
BUSMASTER needs to be stopped and started again to get 'normal active'. If 
we use another tool which you maybe also know the communication gets 
'normal active' w/o restarting. Is there something we have to set inside 
the options or is this a normal behavior?

Many thx in advance,
Robert!

ETAS-Nithin, 2015-10-26:
Hello Robert,

Since we do not have real ECUs to try this out, this issue was not exactly reproducible.
But in our test set-up, while running two instances of BUSMASTER where-in instance-1 of BUSMASTER acts as an ECU (Cold start node) and instance-2 as listener node, resetting instance-1, the instance-2 comes into "normal active" without restarting.

Regards
Nithin

RoSchCo, 2015-10-30:
Hello Nithin!

We have found out following. If  both nodes 

have 'sync node' the restart works. With 'Off' we see the bus before the reset but not after it even the bus is in "normal active".  Is this normal?

Also 2 other topics we have found within FR;
- If the Trace Window is not in Overwrite Mode and the first page is written the first (older) messages on the bus are deleted.
- After setting filter inside the Hardware Settings we are not able to see other messages although we switch off the filtering. Do we have to deactivate something in addition?

Many thx for your support and have a nice weekend in advance,
Robert!

Nithin Nath ---26.10.2015 12:14:09---Hello Robert, Since we do not have real ECUs to try this out, this issue was not exactly reproducibl