rbei-etas/busmaster #841: J1939 Message Handles are not executed.

NorseMN, 2015-09-16:
I have installed 2.6.3 and can't get any message handlers to execute.
I created Message handlers for PGNID, PGNName and PGN_ALL.
I can clearly see in the Message window that my target message is being received.
All handlers are enabled.
Each of the message handlers contains only a Trace instruction.
Nothing appears in the Trace Window.
What am I doing wrong?

RBEI-Prathiba, 2015-09-23:
Hi,
Currently, destination address is considered for broadcast message. Hence, please specify the 'Preferred Address' as '14' for PGN 0xFF14 while creating node in the Node Simulation and check if the handler is executing.

NorseMN, 2015-09-23:
I think BusMaster is failing to recognize a broadcast proprietary message.
The 29-bit identifier used in J1939 is structured in the following way.
Priority = 3 bits
Reserved = 1 bit
Data page = 1 bit
PDU Format (PF) = 8 bits
PDU Specific (PS) = 8 bits
Source Address (SA) = 8 bits

In general:
-  When PS = FF, it's a broadcast message.
-  Otherwise, it's (mostly) a destination specific message and PS is a destination address (DA)

But, there are also proprietary messages that are either destination specific or broadcast.
-  When PF = FE, the message is a destination specific proprietary message.
-  When PF = FF, it's a proprietary broadcast message and PS is a message sub-ID.

The message I'm trying to handle is a proprietary broadcast message.
So the 14 is simply a message sub-ID value, and NOT a destination address.

As far as changing the preferred Node address, I'm puzzled.
It seems that I can only set this address when I create a node.
Then, the only way to change it is to delete the node and create a new node.
When I do that, I lose all of the node definition.
It's a lot of work to recreate each handler function manually.
I think I saw an image of the Node Details screen in you manual that shows the Node Name and Node address as editable fields on this screen,
But for me, they are only displayed values.
Thanks for your help.