rbei-etas/busmaster #842: Reading messages in Visual Basic

StefGeukens, 2015-09-18:
Hello,

I'm trying to make a connection between Visual Basic and BUSMASTER. I was able to connect and send messages to BUSMASTER. But reading messages from BUSMASTER seems difficult.

I tried to use CAN_MonitorApp._IAppEvents and .OnMessageReceived(CAN_RX) but I can’t get it to work. I get errors like, create new object, but I can't or Null reference unhandled.

Is it possible to get a small example to get me going?

Thanks for the help.

RBEI-Prathiba, 2015-09-18:
Hi,
Please refer Examples folder in BUSMASTER installation folder for sample VC++ COM client application which receives data from BUSMASTER. For more details, please refer #383

StefGeukens, 2015-09-18:
Hi,
I already looked at the examples and the posts about the VC++, but I'm working in visual basic and I can't compile the provided code.
Is it possible to recieve a message over a COM interface?
If the answer is yes,
Is it possible to get a visual basic 2010 express example implementing this function using the CAN_MonitorApp as reference (part of the BUSMASTER.exe)?

ones1, 2016-08-17:
Hi, @StefGeukens 
How did you managed to configurate VB to send messages? 
I am just trying the same but I can't get to success...
cheers