rbei-etas/busmaster #843: are admin rights essential for running bustmaster ?

BEG-Katzenberger, 2015-09-21:
Some months ago I had a discussion with Mr. Lorenz about the admin rights busmaster needs for every start. After some tests he came to the conclusion that there is no really need for that and therefore the UAC stuff could be removed in one of the next releases.

So can you please give feedback if this assumption is still correct ?

It would be great help for daily usage because due to company security policy the admin rights must be requested each time ...

Thanks !

RBEI-Prathiba, 2015-09-28:
Hi,
We shall perform some tests with out admin rights and confirm you on the same.

eder-s, 2015-11-03:
Hi,
this is the same problem I reported in #775. It would be great if this would be fixed in one of the next releases as we have the same problems as BEG in the company I work for.