rbei-etas/busmaster #846: support of external editor for node simulation source code ?

BEG-Katzenberger, 2015-09-24:
Using the build-in node simulation editor is quite horrible: no auto indent, source highlighting and other uncountable features which makes coding so comfortable.

I tried to use Eclipse for editing but it seems that the build-in editor is not able to handle a formatted code file.

So please improve the the build-in code editor features significantly or give the user the possibility to choose his favourite editor ...

nostar, 2015-09-24:
The built in editor is certainly not code worthy, but I use external editors with no problem.  I just create a new file, add all of the handlers I think I may need, then save and close the editor.  From then on I always use an external editor.  It's important to keep all of the busmaster generated comments in place though and create new ones around new handlers, for example, if you decide not to create new handlers in the built in editor.  My code editors of choice is Notepad++ for windows and Puma for linux.

BEG-Katzenberger, 2015-09-25:
Using an other editor after creating the "frame" is not a problem - you are right.
But changing this frame afterwards is not so easy. 

For example changing the time of an existing timer handler needs a manual adaptation in the corresponding *.def file by renaming the handler function name.
This took me several hours because the dll is generated without errors but could not be loaded from BusMaster ...