rbei-etas/busmaster #856: VN8970 HW not detected in BUSMASTER

Snigda, 2015-10-29:
We tried using below BUSMASTER version with VN8970 hardware which was procured without a CANalyzer license since we were informed that BUSMASTER does not require vector license.
But while trying to connect, hardware is not getting detected i.e; only virtual channel 1&2 are visible.

[image]

nostar, 2015-10-29:
When you say VN8970 you mean the VN891x base unit with the VN8970 plugged into it right?  Thats a complete realtime computer system running CANoe/CANalyzer via its own RTOS.  We have a couple of these units as well, and essentially when up and running, 2 machines running 2 separate instances of the CANoe/CANalyzer software are communicating back n forth.  I dont see that ever being supported by busmaster.

The Vector XL driver for busmaster supports all of the 'passive' vector hardware.  I use it regularly with PCI(CanBoardXL), PCMCIA(CanCardXL), and USB(CanCaseXL, VN16xx) hardware.

RBEI-Prathiba, 2015-11-17:
Fix is available with v2.6.4

closed 2015-11-17 by RBEI-Prathiba