rbei-etas/busmaster #867: "build node" failed in version 2.6.4

mibcat, 2015-11-17:
Hi !

The new Busmaster version gives me some undefined references:
```
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xcf): undefined reference to `__gxx_personality_sj0'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xfd): undefined reference to `_Unwind_SjLj_Register'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x178): undefined reference to `_Unwind_SjLj_Resume'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x186): undefined reference to `_Unwind_SjLj_Unregister'
collect2: ld gab 1 als Ende-Status zurück
mingw32-make.exe: *** [DllFile.dll] Error 1
```
Same cpp file runs fine with version 6.1.3
Any idea to solve this problem ?

Best regards,
Michael

RBEI-Prathiba, 2015-11-18:
Hello Michael,
Please let us know the version of MinGW you are using by executing 'gcc -v' in the command prompt.
BusMaster v2.6.4 supports 'MinGW TDM1 GCC v4.8.1' which can be downloaded from the below link:
http://sourceforge.net/projects/tdm-gcc/files/TDM-GCC%20Installer/Previous/1.1309.0/tdm-gcc-4.8.1.exe/download
For installation steps, please refer help 'Additional Installation->MinGW Installation using TDM-GCC Installer'

mibcat, 2015-11-20:
Hi Prathiba,

the installed version is **4.6.2**:
```
C:\WINDOWS\system32>gcc -v
Es werden eingebaute Spezifikationen verwendet.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=c:/mingw/bin/../libexec/gcc/mingw32/4.6.2/lto-wrapper.exe
Ziel: mingw32
Konfiguriert mit: ../gcc-4.6.2/configure --enable-languages=c,c++,ada,fortran,objc,obj-c++ --disable-sjlj-exceptions --with-dwarf2 --enable-shared --enable-libgomp --disable-win32-registry --enable-libstdcxx-debug --enable-version-specific-runtime-libs --build=mingw32 --prefix=/mingw
Thread-Modell: win32
gcc-Version 4.6.2 (GCC)
```
I'll try the suggested version and come back with a feedback ...

mibcat, 2015-11-20:
Unfortunately there is no improvement with version 4.8.1

updated MinGW version:
```
C:\Windows\System32>gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=c:/tdm-gcc-32/bin/../libexec/gcc/mingw32/4.8.1/lto-wrapper.exe
Target: mingw32
Configured with: ../gcc-4.8.1/configure --prefix=/mingw --host=mingw32 --build=mingw32 --without-pic --enable-shared --enable-static --with-gnu-ld --enable-lto --enable-libssp --disable-multilib --enable-languages=c,c++,fortran,objc,obj-c++,ada --disable-sjlj-exceptions --with-dwarf2 --disable-win32-registry --enable-libstdcxx-debug --enable-version-specific-runtime-libs --with-gmp=/usr/src/pkg/gmp-5.1.2-1-mingw32-src/bld --with-mpc=/usr/src/pkg/mpc-1.0.1-1-mingw32-src/bld --with-mpfr= --with-system-zlib --with-gnu-as --enable-decimal-float=yes --enable-libgomp --enable-threads --with-libiconv-prefix=/mingw32 --with-libintl-prefix=/mingw --disable-bootstrap LDFLAGS=-s CFLAGS=-D_USE_32BIT_TIME_T
Thread model: win32
gcc version 4.8.1 (GCC)
```
still the same Busmaster error message:
```
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xcf): undefined reference to `__gxx_personality_sj0'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xfd): undefined reference to `_Unwind_SjLj_Register'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x178): undefined reference to `_Unwind_SjLj_Resume'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x186): undefined reference to `_Unwind_SjLj_Unregister'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [DllFile.dll] Error 1
```

mickhh, 2015-11-20:
HI,

build node is successful after installing tdm gcc v4.8.1

Thanks

mibcat, 2015-11-20:
Could you please provide a screen-shot of **'gcc -v'** in the command prompt - Thanks !

mickhh, 2015-11-22:
gcc -v screen-shot

[image]

mibcat, 2015-11-22:
Thanks. Obviously some switches are different but to be honest I've no idea how setup these (correctly) ...

RBEI-Prathiba, 2015-11-23:
Hello Michael,
Please try choosing 'TDM-GCC Recommended, C/C++' component while installation in 
'Choose Components' page.
Please uninstall the existing version before installing new MinGW. 
We shall have WebEx if this doesn't work.

mibcat, 2015-11-23:
Hi Prathiba,

it works now - thanks a lot!

I think the key to happiness was **choosing 'TDM-GCC Recommended, C/C++' component**.
In the Busmaster help file the Additional Installation guide shows here **MinGW Stable, C/C++**.

And this is the version info from gcc:
```
C:\Work\Programme\Console2>gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=C:/TDM-GCC-32/bin/../libexec/gcc/mingw32/5.1.0/lto-wrapper.exe
Target: mingw32
Configured with: ../../../src/gcc-5.1.0/configure --build=mingw32 --enable-languages=ada,c,c++,fortran,lto,objc,obj-c++ --enable-libgomp --enable-lto --enable-graphite --enable-libstdcxx-debug --enable-threads=posix --enable-version-specific-runtime-libs --enable-fully-dynamic-string --enable-libstdcxx-threads --enable-libstdcxx-time --with-gnu-ld --disable-werror --disable-nls --disable-win32-registry --disable-symvers --enable-cxx-flags='-fno-function-sections -fno-data-sections -DWINPTHREAD_STATIC' --prefix=/mingw32tdm --with-local-prefix=/mingw32tdm
--with-pkgversion=tdm-1 --enable-sjlj-exceptions --with-bugurl=http://tdm-gcc.tdragon.net/bugs
Thread model: posix
gcc version 5.1.0 (tdm-1)
```

neliobnjr, 2021-11-05:
Guys please, help.

`Using built-in specs.
COLLECT_GCC=C:\TDM-GCC-64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/TDM-GCC-64/bin/../libexec/gcc/x86_64-w64-mingw32/10.3.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../../../src/gcc-git-10.3.0/configure --build=x86_64-w64-mingw32 --enable-targets=all --enable-languages=ada,c,c++,fortran,jit,lto,objc,obj-c++ --enable-libgomp --enable-lto --enable-graphite --enable-cxx-flags=-DWINPTHREAD_STATIC --disable-build-with-cxx --disable-build-poststage1-with-cxx --enable-libstdcxx-debug --enable-threads=posix --enable-version-specific-runtime-libs --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts=yes --disable-libstdcxx-pch --enable-libstdcxx-threads --enable-libstdcxx-time=yes --enable-mingw-wildcard --with-gnu-ld --disable-werror --enable-nls --disable-win32-registry --enable-large-address-aware --disable-rpath --disable-symvers --prefix=/mingw64tdm --with-local-prefix=/mingw64tdm --with-pkgversion=tdm64-1 --with-bugurl=https://github.com/jmeubank/tdm-gcc/issues
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.3.0 (tdm64-1)`

neliobnjr, 2021-11-05:
[image]