rbei-etas/busmaster #868: Can not see useful hardware id for PCAN dongles

sebnil, 2015-11-20:
In 2.6.0 in Hardware selection I could see the hardware ID. Now in 2.6.4 I only see some other random number. I guess it is some kind of OS device hook number. 

Example:
2.6.4:
PCAN-USB Driver Id 14
2.6.0:
PCAN-USB Device ID - 24h

Configured in PCAN hardware is the 24h number.

Anyway, please revert this change. This bug unfortunatly makes Busmaster unuseble for me.

RBEI-Prathiba, 2015-11-20:
Hi,
From v2.6.3 we have updated Hardware Description to include driver Id which is retrieved from hardware instead of device Id. We could connect and transmit via peak hardware. Please let us know the issue you are facing.

sebnil, 2015-11-25:
The dongles work. No problem there. Problem is that I connect all my dongles via a USB-hub. The dongle hardwares are hidden. Every dongle is given an ID by me in the peak hardware configuration. I assign them numbers like 0x20, 0x21, 0x22 and so on. When I assign them to channels in Busmaster I need to know which dongle is which. Before this was no problem. Now it seems to depend on which order the dongle hardware is recognized. Purely random.

Provide both hardware identifications and the problem is solved.

See screenshot provided in this forum thread:
http://www.peak-system.com/forum/viewtopic.php?f=9&t=305

sebnil, 2016-04-22:
Please update the tag from support to something else. I really do not think this is a support ticket, but rather a real issue.

sebnil, 2018-05-31:
Anything new on this issue? I would very much like to use the new versions of busmaster but am forced to use 2.6.2 because of this bug.

Thank you for a very good software overall.

sebnil, 2019-02-07:
Could you please increase priority on this issue? I like the idea of updating from 2.6.2 to the new version of Busmaster.