rbei-etas/busmaster #872: BUSMASTER listen only mode

hfabian, 2015-11-25:
hello,

I want to use Busmaster for logging in an Listen only mode.
Busmaster should not send an ack or error.
How can I do this?

Thanks for your help.

RBEI-Prathiba, 2015-11-25:
Hi,
We understand that you want to use BUSMASTER for bus monitoring.
Please follow the below steps:
1. Configure Logging using menu 'CAN->Logging->Configure'
2. Enable Logging using menu 'CAN->Logging->Activate'
3. Connect to the hardware

For more details, please refer help section 'CAN->Configuration->Log and Replay'

hfabian, 2015-11-25:
hi,

the problem is, that when I use BUSMASTER for logging (or viewing), that when a Frame comes in, BUSMASTER send a Acknowledge. 
I want to see the communication of two CAN nodes. In the logfile I can not see, when the second node wakes up, when BUSMASTER sends the Acknowledge before.

Can I suppress sending Acknowledge and Errors in BUSMASTER?

In the help section, I can not find anything.