rbei-etas/busmaster #873: msys64/mingw32 didn´t work as compiler

mgiaco, 2015-11-26:
Today i want to try MSYS2 with MinGW32 as compiler for the Node Simulation.

But I get this error:

c:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/5.2.0/../../../../i686-w64-mingw32/bin/ld.exe: cannot find -lin
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [DllFile.dll] Error 1

Has anyone an Idea for what lib -lin  stays?

thx

mgiaco, 2015-12-16:
``` shell
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xcf): undefined reference to `__gxx_personality_sj0'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xfd): undefined reference to `_Unwind_SjLj_Register'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x178): undefined reference to `_Unwind_SjLj_Resume'
C:\Program Files (x86)\BUSMASTER_v2.6.4\SimulatedSystems\OBJ\libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x186): undefined reference to `_Unwind_SjLj_Unregister'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [DllFile.dll] Error 1
```

So the Problem with mingw32 and MSYS2 is the used Exception Handling in the libWrapper_CAN.a. So Is there a way to build this Lib myself?

thanks
mathias

RBEI-Prathiba, 2015-12-18:
Hello mathias,
BusMaster v2.6.4 supports 'MinGW TDM1 GCC v4.8.1' which can be downloaded from the below link:
http://sourceforge.net/projects/tdm-gcc/files/TDM-GCC%20Installer/Previous/1.1309.0/tdm-gcc-4.8.1.exe/download
Please uninstall the existing version, before installing the new MinGW.
For installation steps, please refer BusMaster User Manual 'Additional Installation->MinGW Installation using TDM-GCC Installer'  available in the link "https://raw.githubusercontent.com/rbei-etas/busmaster-documents/master/help.pdf"

mgiaco, 2015-12-21:
Hello,

Thaks for your help but I already know that. For some other projects I use the MSYS2 mingw32 and mingw64 GCC. So my Problem ist I need an extra TDM GCC install for the Busmaster. Therefore, I am looking forward to a libWrapper_CAN.a MSYS2 compiled library. 

MSYS2 is very popular - So I think it would be a good idea to switch to this toolchain in the future.
Can I build the libWrapper_CAN.a by myself?

thx
Mathias

closed 2017-09-06 by mgiaco