rbei-etas/busmaster #874: DBF files do not support Float32/64 data types

jarheck, 2015-12-02:
When creating an new DBF file or converting a DBC file I noticed there is not an option for choosing an IEEE floating point number.

RBEI-Prathiba, 2015-12-28:
Currently BusMaster supports data types only for Raw values. Physical values will be calculated based on the offset, factor and Signal Length etc taken from database.

closed 2015-12-28 by RBEI-Prathiba

jarheck, 2015-12-29:
Yes but there is no way to calculate the physical value of a Floating point number (CANalyzer, etc support this option to view physical values) as this is an allowed datatype in a DBC file. It would be a nice improvement in the future to support this data type.

TheOtherMarcus, 2023-10-20:
This should be reopened. Busmaster doesn't display physical float values correctly when DBF is converted from DBC.