rbei-etas/busmaster #887: Build fails

nostar, 2015-12-18:
Trying to build Debug using VS 2012 ultimate and cmake 3.4.1:

Build FAILED.

"C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln" (default target) (1) ->
"C:\Users\D\Downloads\busmaster\Sources\build\Application\BUSMASTER.vcxproj.metaproj" (default target) (5) ->
"C:\Users\D\Downloads\busmaster\Sources\build\EXTERNAL\DmGraph\DMGraph.vcxproj.metaproj" (default target) (20) ->
"C:\Users\D\Downloads\busmaster\Sources\build\EXTERNAL\DmGraph\DMGraph.vcxproj" (default target) (21) ->
(ClCompile target) ->
  c:\users\d\downloads\busmaster\sources\external\dmgraph\stdafx.h(13): warning C4005: '_CRT_SECURE_NO_WARNINGS' : macro redefinition [C:\Users\D\Downloads\busmaster\Sources\build\EXTERNAL\DmGraph\DMGraph.vcxproj]

"C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln" (default target) (1) ->
"C:\Users\D\Downloads\busmaster\Sources\build\Application\BUSMASTER.vcxproj.metaproj" (default target) (5) ->
"C:\Users\D\Downloads\busmaster\Sources\build\BusEmulation\BusEmulation.vcxproj.metaproj" (default target) (6) ->
"C:\Users\D\Downloads\busmaster\Sources\build\Utility\Utils.vcxproj.metaproj" (default target) (9) ->
"C:\Users\D\Downloads\busmaster\Sources\build\Utility\Utils.vcxproj" (default target) (10) ->
(ClCompile target) ->
  C:\Users\D\Downloads\busmaster\Sources\Utility\UtilFunctions.cpp(126): error C2440: 'initializing' : cannot convert from 'const char' to 'ATL::CStringT<BaseType,StringTraits>' [C:\Users\D\Downloads\busmaster\Sources\build\Utility\Utils.vcxproj]
  C:\Users\D\Downloads\busmaster\Sources\Utility\UtilFunctions.cpp(127): error C2440: 'initializing' : cannot convert from 'char' to 'ATL::CStringT<BaseType,StringTraits>' [C:\Users\D\Downloads\busmaster\Sources\build\Utility\Utils.vcxproj]

"C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln" (default target) (1) ->
"C:\Users\D\Downloads\busmaster\Sources\build\Application\BUSMASTER.vcxproj.metaproj" (default target) (5) ->
"C:\Users\D\Downloads\busmaster\Sources\build\EXTERNAL\DmGraph\DMGraph.vcxproj.metaproj" (default target) (20) ->
"C:\Users\D\Downloads\busmaster\Sources\build\EXTERNAL\DmGraph\DMGraph.vcxproj" (default target) (21) ->
  C:\Users\D\Downloads\busmaster\Sources\EXTERNAL\DmGraph\StdAfx.cpp(9): fatal error C1083: Cannot open include file: 'statreg.cpp': No such file or directory [C:\Users\D\Downloads\busmaster\Sources\build\EXTERNAL\DmGraph\DMGraph.vcxproj]

RBEI-Prathiba, 2015-12-28:
Please use configuration mode as 'busmaster_debug' to use it in debug mode.

nostar, 2015-12-28:
changing the line in build.bat to MSBuild "BUSMASTER.sln" /property:Configuration=busmaster_debug

Now I get:

$ ./build.bat
Using MSBuild found in C:\Windows\Microsoft.NET\Framework\v4.0.30319
Using CMake found in C:\Program Files\CMake\bin
-- The C compiler identification is MSVC 17.0.61030.0
-- The CXX compiler identification is MSVC 17.0.61030.0
-- Check for working C compiler using: Visual Studio 11 2012
-- Check for working C compiler using: Visual Studio 11 2012 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Visual Studio 11 2012
-- Check for working CXX compiler using: Visual Studio 11 2012 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found BISON: C:/Users/D/Downloads/busmaster/Tools/bison/bison.exe
-- Found FLEX: C:/Users/D/Downloads/busmaster/Tools/flex/flex.exe
-- Found GETTEXT: C:/Users/D/Downloads/busmaster/Sources/EXTERNAL/GetText/lib/intl.lib
-- Found ICONV: C:/Users/D/Downloads/busmaster/Sources/EXTERNAL/iconv/lib/iconv.lib
-- Found LIBXML2: C:/Users/D/Downloads/busmaster/Sources/EXTERNAL/libxml2/lib/libxml2.lib
CMake Warning (dev) at CMakeLists.txt:105 (link_directories):
  This command specifies the relative path

    ATL_PATH-NOTFOUND/lib/ATL/i386

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:105 (link_directories):
  This command specifies the relative path

    MFC_PATH-NOTFOUND/lib/Mfc/i386

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
CMake Error at NodeSimEx/CMakeLists.txt:104 (add_library):
  Cannot find source file:

    SimSysConfigDetails.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

-- Generating done
-- Build files have been written to: C:/Users/D/Downloads/busmaster/Sources/build
Microsoft (R) Build Engine version 4.0.30319.18408
[Microsoft .NET Framework, version 4.0.30319.18408]
Copyright (C) Microsoft Corporation. All rights reserved.

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
Build started 12/28/2015 4:40:48 PM.
Project "C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln" on node 1 (default targets).
C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln.metaproj : error MSB4126: The specified solution configuration "busmaster_debug|Win32" is invalid. Please specify a valid solution configuration using the Configuration and Platform properties (e.g. MSBuild.exe Solution.sln /p:Configuration=Debug /p:Platform="Any CPU") or leave those properties blank to use the default solution configuration. [C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln]
Done Building Project "C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln" (default targets) -- FAILED.

Build FAILED.

"C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln" (default target) (1) ->
(ValidateSolutionConfiguration target) ->
  C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln.metaproj : error MSB4126: The specified solution configuration "busmaster_debug|Win32" is invalid. Please specify a valid solution configuration using the Configuration and Platform properties (e.g. MSBuild.exe Solution.sln /p:Configuration=Debug /p:Platform="Any CPU") or leave those properties blank to use the default solution configuration. [C:\Users\D\Downloads\busmaster\Sources\build\BUSMASTER.sln]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:00.17