rbei-etas/busmaster #892: Filter

AutologicAU, 2015-12-29:
Filtering for messages on logging or replay is just simply impossible to use, I have been assured that it works but I cannot make it so

RBEI-Prathiba, 2015-12-29:
To apply Filter on log, follow the below steps:
1. Configure filters using 'CAN->Filter Configuration'
2. Apply the filter to log using 'Configure Logging' window. 'Configure Logging' window can be launched via menu 'CAN->Logging->Configure'
3. Enable Filters for logging using 'CAN->Logging->Enable Filters'

Similar way apply the filters for Replay also. For more details please refer help 'CAN->Configuration->Log and Replay' and 'CAN->Enable/Disable Filter'

AutologicAU, 2015-12-29:
Prathiba,

 

What are you doing working, you should be on holiday?

 

I will try again and report, no good thus far

 

 

All the best

 

Ian

Ian Crane BSc RSV

Autologic

BMW Specialists

www.autologic.com.au

Australia

613 9563 8420

613 (0)409 578575

AutologicAU, 2015-12-29:
Prathiba,

 

I have spent hours trying to get a replay filter going on a log file using two CAN I.D.s….cannot get it to work…

 

This must have a revision by way of the filter dialog, it is just impossible.

 

Have a happy new year won’t you?

 

 

 

All the best

 

Ian

Ian Crane BSc RSV

Autologic

BMW Specialists

www.autologic.com.au

Australia

613 9563 8420

613 (0)409 578575

AutologicAU, 2016-01-01:
Prathiba,

 

I managed to get the filter to work.

 

Hell if I know how.

 

I will clear my entire session and attempt to set it up again.

 

Will advise what was the missing link once I work it out

 

Happy New Year and thanks

 

 

All the best

 

Ian

Ian Crane BSc RSV

Autologic

BMW Specialists

www.autologic.com.au

Australia

613 9563 8420

613 (0)409 578575

AutologicAU, 2016-01-03:
I have now been able to set filters twice.

The missing step in my setup was to go to replay configeration-----> filters---->filter Selection dialog---->add [filters]

closed 2016-02-01 by RBEI-Saravanan