rbei-etas/busmaster #895: Time format

AutologicAU, 2015-12-30:
Busmaster expresses time in the format hh:mm:ss:ms.

When exporting this to excel it is next to impossible to deal with

I suggest that within busmaster and for export that the internal time format should be hh:mm:ss.ms

gxurma, 2015-12-30:
2015-12-30 06:10 keltezéssel, AutologicAU írta:

> Busmaster expresses time in the format hh:mm:ss:ms
>
> When exporting this to excel it is next to impossible to deal with
>
> I suggest that within busmaster and for export that the internal time
> format should be hh:mm:ssms

I would suggest to use hh:mm:ss.msms

Also I would suggest to be able to log the date, and the system time, 
not the continuous time. Date should be importable by excel.

AutologicAU, 2015-12-30:
Yes correct, thank you

AutologicAU, 2015-12-30:
In excel the format will need to be hh:mm:ss.0000

AutologicAU, 2015-12-31:
I have written a piece of VB code which will convert the Busmaster time format to HH:mm:ss.ms.

I'm no programmer but logical and it was not easy to get excel to present the time in the right format.

That done if I subtract an earlier message time from a later message time to get the transmission interval, as yet I have not got that working.

This issue may be greater than simply changing the format.

Perhaps we should look at creation of a calculated transmission interval between messages and between messages when filtering is on so that we can see the interval of transmission on a particular can ID.