rbei-etas/busmaster #897: Member Forum

AutologicAU, 2015-12-31:
I think it would be a good idea if those interested who are registered contributors could discuss Busmaster and CAN or other bus issues by way of a forum.

This may better equip users and also to accelerate evolution of the ideas for the product through open discussion.

nanos gigantum humeris insidentes

krish97git, 2019-09-26:
Anyone tried to interface busmaster with CAN module like mcp2515 and arduino?
Does it work? If so please share the procedure

cbindi, 2021-03-12:
Hello all,

where I can looking for a tutorial in order to use the node simulation busmaster feature?

I don't understand how I'ld be able to use the dbf in order to send specified messages. 

If I click the right mouse button I see te popup where I see messages and signal but clicking on this labels, an irrilevant code is add to the functions.

Is there a tutorial that I could follow?

I tried t use visualstudio code in order to check the data structures used in busmaster, but I received a lot of error message on the defines (like GCC_EXTERN and so on...) that they prevent the right code interpretation.
There is someone that could help me...

thank you s muche in advance!

CB