rbei-etas/busmaster #898: Busmaster-unable to create new function editor file

Khanalb, 2016-01-02:
Hi,
I am using busmaster v2.6.4 and I am having trouble creating new function editor file in CAN.
When I do Can>>Node Simulation>>Configure then add a node, I don't see an 'Add New File...' option when I right click on the node. 

I am using busmaster in 'passive' mode to setup node behavior until my hardware arrives. Is this the reason why I cannot create new function editor?

please help.

RBEI-Prathiba, 2016-01-04:
Hi,
To create new .cpp file or to load existing file, please select the file in 'Node Details' dialog using browse button while adding a node.

Khanalb, 2016-01-04:
I created a new .cpp file using notepad and load it into 'Node Details' dialog box. However, when I do that it prompts an error message that says the given file is not created for CAN. Please open the .cpp file created for CAN (pease see image attached). 
How would I create a .cpp file specific to CAN?  
Sorry about the nube question.
I am new to BUSMASTER but I have experience with CANalyzer and CANoe. I feel like I know what I am doing but I have spend couple of hours just trying to create a new file. 
I have done quite a bit of codeing in C but have no experience with c++.
Are nodes coded in C or C++ in BUSMASTER ?
I appreciate your response.
[image]

RBEI-Prathiba, 2016-01-08:
Please do not create .cpp file manually as BusMaster has its own format. To create new .cpp file please specify the file name in the 'File dialog' opened on browse button of 'Node Details' dialog

Rohith1881, 2022-11-25:
[image]
i am getting this error