rbei-etas/busmaster #900: Time since last message with same Identifier.

AutologicAU, 2016-01-03:
Perhaps as part of each message statistic within Busmaster we could record the time since a message with the same Identification was recorded.
Will make dealing with time in an export easy as Excel does not handle time well at all

RBEI-Saravanan, 2016-01-11:
We understand the query is to log the frames with relative time mode which would be relative to same identifier. If this is the case, log provides an option to configure the relative time mode in the Time Mode drop down in Log configuration window.

AutologicAU, 2016-01-11:
Thank you I will investigat that.

 

Happy New Year

 

 

All the best

 

Ian

Ian Crane BSc RSV

Autologic

BMW Specialists

www.autologic.com.au

Australia

613 9563 8420

613 (0)409 578575