rbei-etas/busmaster #901: Node simulation not working in BUSMASTER 2.6.4

akshay29, 2016-01-16:
Hello,

I have installed latest version of MinGW for BUSMASTER. Though it got installed properly, I am still not able to use node simulation.On building the node I get a message as mingw32-g++.exe is not recognised as an internal or external command,operable program or batch file . 

For your reference I am attaching the screenshots of the error message as well as the installed compiler version.

Kindly help me out in resolving the issues.

[image]
[image]

RBEI-Prathiba, 2016-01-18:
Hello,
BusMaster v2.6.4 supports 'MinGW TDM1 GCC v4.8.1'. 
Use 'Download MinGW' option and ‘User Manual’ option from the main page 'http://rbei-etas.github.io/busmaster/' to download and install the same.

Please refer section ‘Additional Installation->MinGW Installation using TDM-GCC Installer’ in the user manual for the installation steps.

akshay29, 2016-01-18:
Hello,
I tried installing MinGW TDM1 GCC v4.8.1 but got cannot download from repository message.Hence,
I installed MinGW TDM1 GCC v5.0.1 successfully as seen in the screenshot.

Isnt the MinGW TDM1 GCC v5.0.1 compatible with latest BUSMASTER version?

Does it works only with MinGw v4.8.1?

Thanks and Regards,
Akshay Kulkarni

RBEI-Srinibas, 2016-01-18:
Hello,

Currently BusMaster v2.6.4 supports 'MinGW TDM1 GCC v4.8.1' only , which can be downloaded from the below link:
http://sourceforge.net/projects/tdm-gcc/files/TDM-GCC%20Installer/Previous/1.1309.0/tdm-gcc-4.8.1.exe/download

For installation steps, please refer BusMaster User Manual 'Additional Installation->MinGW Installation using TDM-GCC Installer' available in the link "https://raw.githubusercontent.com/rbei-etas/busmaster-documents/master/help.pdf"

Thanks,
Srinibas

akshay29, 2016-01-18:
Hello,

As mentioned earlier, I  tried Installing MinGW TDM1 GCC v4.8.1'  but it gave an error(refer the attachment).

Regards,

Akshay Kulkarni

http://sourceforge.net/projects/tdm-gcc/files/TDM-GCC%20Installer/Previous/1.1309.0/tdm-gcc-4.8.1.exe/download

Download TDM-GCC MinGW Compiler from SourceForge.net<http://sourceforge.net/projects/tdm-gcc/files/TDM-GCC%20Installer/Previous/1.1309.0/tdm-gcc-4.8.1.exe/download>
sourceforge.net
GCC for 32-bit and 64-bit Windows with a real installer & updater

RBEI-Srinibas, 2016-01-20:
Hi,

Could you please check that first two steps [ Refer page 257 in help.pdf ]  of MinGW installation are followed.

step1-  check box for  "check for updated files on TDM-GCC Server" must be unchecked.
step2-  select edition.

Thanks,
Srinibas

akshay29, 2016-01-20:
Hi,

Thank you very much Srinibas. Finally it works :)

I would like to get a clarification  related to CAN-FD.
Have you guys planned a release or a milestone for CAN-FD support with Vector hardware in BUSMASTER?

Thanks and Regards,
Akshay