rbei-etas/busmaster #909: LDF editor

bosca1965, 2016-02-02:
Hello,
I'm using busmaster ver 2.64 with Kvaser LIN to spy a LIN network, but it seems not all work properly:

- LDF editor doesn't start
- The LIN message window shows only the ID message , no data byte and checksum (with label "classic" instead of "enhanced", I'm using LIN 2.1)
A question, does bus master decode LIN messages? (the signal name and encoding of signal as defined on LDF file is managed?)

RBEI-Srinibas, 2016-02-04:
Yes, Busmaster can decode LIN messages from LDF file.
Please send the LDF file to busmaster@in.bosch.com .We will analyze and let you know.

Thanks,
Srinibas

jtrzhangjie, 2024-06-12:
Hello,
I'm using busmaster ver 2.64 with Kvaser LIN to spy a LIN network, but it seems not all work properly:

· LDF editor doesn't start