rbei-etas/busmaster #916: Transmission of J1939 message

snehasethi, 2016-02-23:
Hi Team,

I am unable to get the broadcast option in the Msg type drop down in J1939 Tx Message window. Kindly let me know what needs to be done.

RBEI-Srinibas, 2016-03-04:
message type will be determined by the PF Value in PGN. For  more details ,Please refer #205.

astrand, 2016-07-02:
Busmaster does not work correctly. For example, when trying to send PGN=0xfeec, Type=Broadcast, From=0xfe, To=0xf3, DLC=18, Prio=7, Data=0102030405060708090a0b0c0d0e0f10112a, then Busmaster will create a TPCM with data bytes:

20 12 00 03 ff f3 fe 00

ie for a PGN fef3, instead of feec. Setting To=0xec will, as a workaround, give a correct TPCM, but then the TPDT will still be wrong and have an ID with DA=0xec. J1939/21 is pretty clear about what DA should be set to: "Global (DA = 255) for TP.CM.BAM data transfers"