rbei-etas/busmaster #920: Incrementing CAN signal value through node simulation

akshay29, 2016-03-02:
Hello,

I want to increment the value of a CAN signal based on timer. For instance, If the timer is running for 100 ms then for the first 100ms the value should be 1,for 200ms it should be 2 and so on. For this I have made use of a modulo operator but the value is not getting incremented as desired.Kindly find my code snippet below : 

I have defined variable 'i' in the global variables section.

void OnTimer_timer_100ms_100( )
{
Message1 sMsgStruct;
 
sMsgStruct.m_sWhichBit.Sig1= i % 100 ;
++i;
SendMsg((*(STCAN_MSG*)&sMsgStruct));
 }

void OnBus_Connect()
{
StartTimer("timer_100ms", 100);
}
void OnBus_Disconnect()
{
StopTimer("timer_100ms");
}

Kindly let me know what changes need to be done.

Regards,
Akshay

nostar, 2016-03-03:
I think that if you used Trace() to see i you would find that i is
incrementing and that there is probably something wrong with your can
message.  Instead of using a database, trying using a raw STCAN_MSG and
populate the first byte, like the code below.  Also make sure that i is
defined as a compatible type.  I add ```#include <inttypes.h>``` to all of
my simulations and use uint8_t for single byte values.

Also note that as of 2.6.4 all handlers are started automatically on
connect.

```
/* This file is generated by BUSMASTER */
/* VERSION [1.1] */
/* BUSMASTER VERSION [2.6.4] */
/* PROTOCOL [CAN] */

/* Start BUSMASTER include header */
#include <Windows.h>
#include <CANIncludes.h>
#include <inttypes.h>

/* End BUSMASTER include header */

/* Start BUSMASTER global variable */
uint8_t i = 0;
/* End BUSMASTER global variable */

/* Start BUSMASTER Function Prototype  */
GCC_EXTERN void GCC_EXPORT OnTimer_timer_100ms_100( );
/* End BUSMASTER Function Prototype  */

/* Start BUSMASTER Function Wrapper Prototype  */
/* End BUSMASTER Function Wrapper Prototype  */

/* Start BUSMASTER generated function - OnTimer_timer_100ms_100 */
void OnTimer_timer_100ms_100( )
{
STCAN_MSG sMsgStruct;
sMsgStruct.m_unMsgID = 0x100;
sMsgStruct.m_ucDLC = 1;
sMsgStruct.m_ucChannel = 1;
sMsgStruct.m_sWhichBit.m_aucData[0] = i % 100;
++i;
Trace("i = %d", i);
SendMsg(sMsgStruct);
}/* End BUSMASTER generated function - OnTimer_timer_100ms_100 */
```

Mazighe, 2016-03-03:
can you please load the hole project
thank you

nostar, 2016-03-07:
[github_920.zip](https://github.com/rbei-etas/busmaster/files/161525/github_920.zip)

RBEI-Srinibas, 2016-03-21:
Hello,
Please also provide DLC and Channel details as shown below.

void OnTimer_timer_100ms_100( )
{
Message1 sMsgStruct;
sMsgStruct.m_ucChannel = 1;
sMsgStruct.m_ucDLC= 8;
sMsgStruct.m_sWhichBit.Sig1= i % 100 ;
++i;
SendMsg(((STCAN_MSG)&sMsgStruct));
}

Regards,
Srinibas