rbei-etas/busmaster #923: Problem transmitting LIN frames > 4 bytes

nostar, 2016-03-24:
Using Peak-CAN PCAN-USB Pro FD hardware, I can transmit slave response frames < 4 bytes with no problem.  The upper 4 bytes always transmit random values.  I have tried every combination of LIN protocol and baud rate with no change.  Note that in the source code I have commented out the word and long types, as I have tried all 3 with the same result on the last 4 bytes.

Example output:
```
***BUSMASTER Ver 2.6.4***
***PROTOCOL LIN***
***NOTE: PLEASE DO NOT EDIT THIS DOCUMENT***
***[START LOGGING SESSION]***
***START DATE AND TIME 24:3:2016 11:54:56:695***
***HEX***
***SYSTEM MODE***
***START CHANNEL BAUD RATE***
***CHANNEL 1 - Protocol Version LIN 2.0 - Hardware PLIN-USB Device ID - 1h - BaudRate 19200 bps***
***END CHANNEL BAUD RATE***
***START OF LDF FILES***
***END OF LDF FILES***
***<Time><Tx/Rx><Channel><ID><DLC><DataBytes><Checksum>***
11:54:56:6755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:56:7755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:56:8755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:56:9755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:0755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:1755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:2755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:3755 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:4756 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:5756 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:6826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:7826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:8826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:57:9826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:0826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:1826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:2826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:3826 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:4827 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:5827 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:6847 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:7847 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:8847 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
11:54:58:9847 Tx 1 0x001 8 12 34 56 78 C0 FC FB 10  0x5F (Enhanced) 
***END DATE AND TIME 24:3:2016 11:54:59:374***
***[STOP LOGGING SESSION]***
```

```
/* This file is generated by BUSMASTER */
/* VERSION [1.1] */
/* BUSMASTER VERSION [2.6.4] */
/* PROTOCOL [LIN] */

/* Start LIN include header */
#include <Windows.h>
#include <LINIncludes.h>

/* End LIN include header */

/* Start LIN global variable */
STLIN_MSG cMsg;
STLIN_MSG sMsg;
/* End LIN global variable */

/* Start LIN Function Prototype  */
GCC_EXTERN void GCC_EXPORT OnTimer_100ms_100( );
/* End LIN Function Prototype  */

/* Start LIN Function Wrapper Prototype  */
/* End LIN Function Wrapper Prototype  */

/* Start LIN generated function - OnTimer_100ms_100 */
void OnTimer_100ms_100( )
{
	cMsg.m_ucMsgTyp = 0;
	cMsg.m_ucChksumTyp = 1;
	cMsg.m_ucDataLen = 8;
	cMsg.m_ucMsgID = 0x01;
	cMsg.m_ucChannel = 1;
	SendMsg(cMsg);

	sMsg.m_ucMsgID = 0x01;
	sMsg.m_ucMsgTyp = 1;
	sMsg.m_ucChksumTyp = 1;
	sMsg.m_sWhichBit.m_aucData[0] = 0x12;
	sMsg.m_sWhichBit.m_aucData[1] = 0x34;
	sMsg.m_sWhichBit.m_aucData[2] = 0x56;
	sMsg.m_sWhichBit.m_aucData[3] = 0x78;
	sMsg.m_sWhichBit.m_aucData[4] = 0x87;
	sMsg.m_sWhichBit.m_aucData[5] = 0x65;
	sMsg.m_sWhichBit.m_aucData[6] = 0x43;
	sMsg.m_sWhichBit.m_aucData[7] = 0x21;
	//sMsg.m_sWhichBit.m_auwData[0] = 0x1234;
	//sMsg.m_sWhichBit.m_auwData[1] = 0x5678;
	//sMsg.m_sWhichBit.m_auwData[2] = 0x8765;
	//sMsg.m_sWhichBit.m_auwData[3] = 0x4321;
	//sMsg.m_sWhichBit.m_aulData[0] = 0x12345678;
	//sMsg.m_sWhichBit.m_aulData[1] = 0x87654321;
	sMsg.m_ucChannel = 1;
	sMsg.m_ucDataLen = 8;
	SendMsg(sMsg);
}/* End LIN generated function - OnTimer_100ms_100 */
```

nostar, 2016-03-24:
This problem appears to be specific to the Peak hardware.  I just tried a vector VN1630 and it works correctly.

nostar, 2016-03-25:
Here is the fix
```
diff --git a/Sources/LIN_PEAK_USB/LIN_PEAK_USB.cpp b/Sources/LIN_PEAK_USB/LIN_PEAK_USB.cpp
index 6d38dd4..8b76a40 100644
--- a/Sources/LIN_PEAK_USB/LIN_PEAK_USB.cpp
+++ b/Sources/LIN_PEAK_USB/LIN_PEAK_USB.cpp
@@ -1792,7 +1792,7 @@ HRESULT CDIL_LIN_PeakUSB::LIN_Send(STLIN_MSG& sLinTxMsg)
             {
                 msg.Direction = dirPublisher;
                 msg.ChecksumType = odChannel.m_linChecksumType;
-                memcpy(msg.Data, itrData->second.ucData, sizeof(itrData->second.m_nLength));
+                memcpy(msg.Data, itrData->second.ucData, itrData->second.m_nLength);
                 dwRet = pfLIN_CalcChecksum(&msg);
             }
             else                                                //Master Task Only
@@ -2144,4 +2144,4 @@ HRESULT CDIL_LIN_PeakUSB::LIN_GetErrorCount(SERROR_CNT& sErrorCnt, UINT nChannel
         hResult = ERR_IMPROPER_STATE;
     }
     return hResult;
-}
\ No newline at end of file
+}
```

RBEI-busmaster, 2016-07-01:
Fixed the issue. Updated with v3.0.0

closed 2016-07-01 by RBEI-busmaster