rbei-etas/busmaster #937: Contribution of Multiple Y-Axis to signal graph

SharmeelaAshwin, 2016-05-06:
I am a Masters student and as part of  my 'communication systems' lab I would like to contribute multiple Y-Axis in signal graphs

ETAS-Nithin, 2016-05-11:
You are most welcome! 
Kindly refer to the contribution guidelines on how to proceed. You may also use this #937 to publish updates on your progress for the benefit of other users in the community. Good luck!

SharmeelaAshwin, 2016-05-11:
Thank you Mr. Nath. I will surely refer the guidelines.

SharmeelaAshwin, 2016-05-13:
I would like to use C# to develop the graph and add it as a plugin to Busmaster. Kindly advice me on the feasibility of the same.Thank you.

RBEI-busmaster, 2016-06-03:
C# can be used for development and to integrate with BusMaster.

ETAS-Nithin, 2016-07-08:
@SharmeelaAshwin , any updates on the progress on the sig-graph enhancement?

SharmeelaAshwin, 2016-07-11:
Hello Mr. Nath,
Sorry for late reply. I was busy coding as I had to finish some thing before I respond. 
Graph is working already. It is able to plot the values. I need to finish it by this week end as my lab presentation will be in next week. There are a few things pending. I am trying my best to finish it on time. Please find the image of the graph below
[image]

SharmeelaAshwin, 2016-07-30:
Hello Mr.Nath,
I am done with my coding and I have already committed the code. Please let me know how do we proceed.