rbei-etas/busmaster #940: Message window interprets values wrong

schnurksel, 2016-05-25:
The Message window interprets the message values wrong while at the same moment the same value is interpreted inside the signal watch window right.
Example: The Raw value is 0x3E0 at the signal window and the message window.
The signal window interprets this value right as as -32 while the message window shows 992.
See attached image.
[image]
The value is a 10 Bit signed value with offset 0. The message windows shows it as an unsigned value (1024 - 32 = 992).

RBEI-VENKAT, 2016-05-27:
We are working on this issue along with the similar issues ( #719 , #750, #756 )

RBEI-busmaster, 2016-07-01:
Fixed the issue. Updated with v3.0.0

closed 2016-07-01 by RBEI-busmaster