rbei-etas/busmaster #949: Crash at application close & Flexray menu not showing

Phreak-Show, 2016-07-02:
Busmaster 3.0 crashes with "Unhandles exception in busmaster. Would you like to save a dump file" everytime I close it, on Win10 x64 and Win8.1 x64.

It also does not display and Flexray-related button, the former 2.6.x did. Is there now support for Vector's flexray hardware, like the 7610?

ETAS-Nithin, 2016-07-07:
Busmaster is currently supported on Win 7 x64. Support for x64 versions of 8.1 and 10 are planned in subsequent releases. Do you get the same behavior even when run in Win7 compatibility mode ?

With 3.0 there was a major architectural over-haul.  This is currently done for LIN & CAN. We plan to extend the same to FlexRay functionalities in the future.

But in case, you would require any further information regarding FlexRay features and availability, please contact busmaster@in.bosch.com.

Phreak-Show, 2016-07-19:
Yes, it's the same behavior when I run it in compatibility mode for 7 and XP.

Szendvics, 2017-02-02:
Same error here.
Dump can be reached at this link:
https://dl.dropboxusercontent.com/u/4410106/BUSMASTER.dmp

GR8DAN, 2017-03-15:
Here's the dump file for the crash on my Win 10 x64 machine:

Microsoft (R) Windows Debugger Version 10.0.14321.1024 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Users\john

doe\AppData\Local\Temp\BUSMASTER.dmp]
User Mini Dump File: Only registers, stack and portions of memory are available

Symbol search path is: srv*
Executable search path is: 
Windows 10 Version 14393 MP (4 procs) Free x86 compatible
Product: WinNt, suite: SingleUserTS
Built by: 10.0.14393.206 (rs1_release.160915-0644)
Machine Name:
Debug session time: Wed Mar 15 01:26:22.000 2017 (UTC + 0:00)
System Uptime: not available
Process Uptime: 0 days 0:00:38.000
................................................................
...............
This dump file has an exception of interest stored in it.
The stored exception information can be accessed via .ecxr.
(14e0.11f8): Access violation - code c0000005 (first/second chance not available)
eax=00000000 ebx=0ba61008 ecx=0ba398d4 edx=ffffffff esi=000002cc edi=0ba60ff8
eip=77def43c esp=015ed3e4 ebp=015ed3f0 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000206
ntdll!NtGetContextThread+0xc:
77def43c c20800          ret     8